moodlebox/moodle-tool_moodlebox #12: Unnecessary calls to exec()

mudrd8mz, 2017-04-11:
Although in this particular case they all are pretty safe, it is always a bit scary to see Moodle plugins to do any of potentially dangerous calls like `exec()`. Especially if there is no real reason for them.

For things like `exec('awk \'/^Hardware/{print $3}\' /proc/cpuinfo');` or even `exec('cat /var/lib/misc/dnsmasq.leases', $leases);`, why not to use native PHP commands like `file_get_contents()`. 

Similarly, I can't see why there must be `exec("touch $reboottriggerfilename");` when there is a PHP wrapper for that http://php.net/manual/en/function.touch.php (by the way, it feels like the path to these triggers could better be explicit).

Was there a special reason why exec() is used instead of the suggested PHP alternatives?

martignoni, 2017-04-12:
Didn't know about touch in PHP. Thanks for this.

And no, there's no special reason. I'll fix it.

martignoni, 2017-04-12:
Some fixes for #12. Closing.

closed 2017-04-12 by martignoni