moodlebox/moodle-tool_moodlebox #13: Incomplete description at moodle.org/plugins

mudrd8mz, 2017-04-11:
The description reads

> After the installation in Moodle, some steps are required to complete on the Raspberry Pi (see below).

There is not below and not steps are described there. I assume you wanted to document the cron job watching for the trigger files?

Additionally, the plugin implicitly expects that some kind of Linux is installed on the Raspberry Pi (which even if is the case in 99%, may or may not be always true). That and all other requirements should be properly documented.

martignoni, 2017-04-12:
Oh, I dumbly thought that the README file was displayed "below". Is it OK to send users to read the README file (where the whole installation process is thoroughly described)?

mudrd8mz, 2017-04-12:
Yes. Additionally you can use the URL https://github.com/martignoni/moodle-tool_moodlebox/blob/master/README.md as the value of the Documentation link to make it even easier.

martignoni, 2017-04-12:
Done, thanks.

closed 2017-04-12 by martignoni