moodlebox/moodle-tool_moodlebox #14: Simplify styles.css by giving the theme actual control

mudrd8mz, 2017-04-11:
In recent Moodle versions, plugins are encouraged to let the theme do the most of the styling. Instead of specifying how widgets look like - e.g. the buttons in your case - it is generally better to just specify the CSS classes and let the theme do the job. This works particularly well with bootstrap based themes like Clean or Boost.

martignoni, 2017-04-12:
Can't find how to override btn-primary class with mform createElement method. Still searching.

mudrd8mz, 2017-04-13:
Not tested, but something like this could help maybe
```
$buttonarray[] = $mform->createElement('submit', 'restartbutton', get_string('restart', 'tool_moodlebox'),
    ['class' => 'btn btn-danger']);
```

martignoni, 2017-04-13:
I tried this, but it strangely append the class to the form-group div around the button, instead of the button itself. Bug?

[image]

mudrd8mz, 2017-04-13:
No, it actually looks like a Boost theme bug. In the Clean theme (which uses the default method of buttons rendering, the class is applied to the `<input>` element, not the wrapper. But, even in Clean, it does not work - because the theme's styles for the buttons override the .btn-danger styles.

This is what I used for testings:
```
<?php

require(__DIR__.'/config.php');
require_once($CFG->libdir.'/formslib.php');

class restartshutdown_form extends moodleform {
    /**
     * Define the form.
     */
    public function definition() {
        $mform = $this->_form;
        $buttonarray = array();
        $buttonarray[] = $mform->createElement('submit', 'restartbutton', 'Restart', ['class' => 'btn btn-danger ']);
        $buttonarray[] = $mform->createElement('submit', 'shutdownbutton', 'Shutdown', ['class' => 'btn btn-danger']);
        $mform->addGroup($buttonarray, 'buttonar', '', array(' '), false);
        $mform->closeHeaderBefore('buttonar');
    }
}

$PAGE->set_url('/f.php');
$PAGE->set_context(context_system::instance());

$f = new restartshutdown_form();

echo $OUTPUT->header();
echo $f->render();
echo $OUTPUT->footer();
```

Anyway, please do not spent too much time on this and feel free to close. Apparently even I thought it would be much more easier than it actually is.

martignoni, 2017-04-14:
Ok, I'll close this for now, and re-open it when Boost is fixed. Thanks David for your tests and help. Much appreciated.

closed 2017-04-14 by martignoni

martignoni, 2017-04-17:
Just opened https://tracker.moodle.org/browse/MDL-58607 on Moodle tracker.