moodlebox/moodle-tool_moodlebox #15: language files

ralf-krause, 2017-04-24:
Hello Nicolas,
with the new version you added the language files for de and es_mx into the package. 

I think this is not a good idea because nobody will be able to change the text strings. A plugin for Moodle should only come with the english text strings. The other strings will come with the language pack. When I add the german language pack de then I get all strings from the official language distribution ... and this language pack also comes with the file moodlebox.php.

Best regards, Ralf

ralf-krause, 2017-04-24:
Hi Nicolas,
I tested the problem again. 

For your update to the plugin version 1.5.1 you changed a string. The string is changed in your locally saved language packs but the string is not changed in the language pack in AMOS. For my test I removed the file .newpassword.sh from the plugin folder. I wanted to get the error message in the moodlebox settings. Moodle uses the string from the official language distribution and not the string from your locally changed file in the plugin version 1.5.1.

Best regards, Ralf

ralf-krause, 2017-04-24:
After changing the german string in AMOS now I get the correct error message in the moodlebox settings.

martignoni, 2017-04-25:
Hi Ralf,
Thanks for your feedback. I did it to be sure that someone installing the plugin will get at least a translation (even if not up-to-date) in all languages, even without making language pack updates. 

I understand that this has a number of drawbacks. And since when installing a new language pack, the strings will be downloaded, I'll pull out the strings of the plugin.

martignoni, 2017-04-25:
Fixed, closing. Please reopen if not OK. Thank you.

closed 2017-04-25 by martignoni

ralf-krause, 2017-04-25:
Hi Nicolas,
if someone installs the plugin then he must get the english language pack ... this is the standard for Moodle and this is the standard for all plugins. If you preinstall the french language pack on your MoodleBox then you will also need the preinstalled french text strings for the MoodleBox plugin.

martignoni, 2017-04-25:
Yes, and this is not a problem, as when you install the lang pack (say french), you get the newest strings of the plugin.

PS. I chose to keep the fr language in the source for historical reasons. :-)