moodlebox/moodle-tool_moodlebox #33: Make restart and shutdown buttons more accessible

martignoni, 2017-12-08:
Idea suggested here: https://moodle.org/mod/forum/discuss.php?d=362704.

Possibilities:
- New block plugin, or
- Current plugin injects the buttons via JS.

(Issue cloned from: https://github.com/martignoni/moodlebox/issues/50.)

martignoni, 2017-12-08:
See https://docs.moodle.org/dev/Output_callbacks#before_footer.

Example in Moodle source code: function tool_usertours_before_footer() in /tool/usertours/lib.php.

martignoni, 2018-10-02:
Don't forget to add an option, so that users can opt-out from having those buttons displayed.

adpe, 2018-11-10:
Sound's interesting. What about a solution, if the button are displayed for site-admins in the navigation drawer instead of a block?

martignoni, 2018-11-10:
Also a possibility. My idea was to display those buttons for the admin only, on __every page__ (in the footer).

adpe, 2018-11-14:
The contact form isn't working on https://moodlebox.net, therefore I contacted you private on https://moodle.org.

martignoni, 2019-03-20:
Tested. Works as described.

martignoni, 2019-04-15:
All done. Closing.

A big thank you to @adpe.

closed 2019-04-15 by martignoni