moodlebox/moodle-tool_moodlebox #38: Enable SSID with UTF-8 chars

martignoni, 2018-03-16:
Allow UTF-8 chars in SSID. See #31.

martignoni, 2018-03-16:
Use `ssid2` instead of `ssid`, e.g. these lines in config file:
```bash
ssid2=P"MoodleBox"
utf8_ssid=1
```

martignoni, 2018-03-18:
Fixing #31 enabled chars with accent or other diacritical mark to be accepted in SSIDs.

This needs some more testing.

martignoni, 2018-03-19:
Complete UTF-8 support is difficult to implement in GUI.

Some references:
- https://github.com/moodlebox/moodle-tool_moodlebox/issues/38

ralf-krause, 2018-03-19:
Hi Nicolas,
I found that a ssid must have less or equal 32 characters. Some people say that a ssid should not have no space and no utf8 characters and no emojis. But it's a nice idea to show ssids with emojis or other signs.

I tested some routers and with german umlauts and some emojis. I do not get any problem with iOS 9 (and up) and with macOS 10.6 (and up). Also I get no problem with Android 7.0.0 but I can't say if a very old Android 4.x will do so ... I don't know what Windows 7 or Windows 10 will do.

But for a MoodleBox we will have a small user group. If something does not work the it's easy to change the ssid.

Ralf

martignoni, 2018-03-19:
> I found that a ssid must have less or equal 32 characters.

Not exactly: SSID must have less or equal **bytes**, and a character can take between 1 and 4 bytes. E.g. this string "𝔐𝔬𝔬𝔡𝔩𝔢𝔅𝔬𝔵" has 9 characters, but 36 bytes!

Thanks for your testing of routers.

martignoni, 2018-03-19:
Found a good solution (see above). Will ship with next version of plugin. More testing needed.

martignoni, 2018-03-19:
Tested, documentation done. Closing.

closed 2018-03-19 by martignoni