moodlebox/moodle-tool_moodlebox #39: Support of RPi 3B+

martignoni, 2018-03-16:
Add detection of Raspberry Pi 3B+.

martignoni, 2018-03-16:
See commits c3715f8 and 526ddb4.

martignoni, 2018-03-16:
Waiting my RPi3B+ to test this.

martignoni, 2018-03-21:
Closing as fixed.

closed 2018-03-21 by martignoni

ralf-krause, 2018-03-21:
Hello Nicolas,
the plugin has the version number 1.10. But in AMOS I can‘t see the text string for the Raspberry Pi 3B+.
Ralf

martignoni, 2018-03-21:
You're right. Though it's present in the release (so you see it in the GUI), it is not present in AMOS. I suspect that AMOS can't interpret correctly the "+" sign in a string key.

Will ask in AMOS forum about this.

ralf-krause, 2018-03-21:
I think you must change the text Id to someting like rpi3b-plus.
$string['rpi3b+'] = 'Raspberry Pi 3 model B+';

martignoni, 2018-03-21:
> I suspect that AMOS can't interpret correctly the "+" sign in a string key.

Indeed.

martignoni, 2018-03-23:
@ralf-krause: I just cut a new release with a fix (see #40). Thanks for the heads up.