moodlebox/moodle-tool_moodlebox #41: Impossible to enter SSIDs with less than 8 bytes

martignoni, 2018-04-29:
Regression added in a6c96f126ca333fb10750165161084c7c7dd0b34.

martignoni, 2018-04-29:
Tested. Works as advertised.

martignoni, 2018-04-29:
Fixed by 30ec469.

closed 2018-04-29 by martignoni

ralf-krause, 2018-08-27:
Hi Nicolas, 
in this issue you fix the problem that nobody could enter a ssid with only one character. This is correct. You can enter any ssid such as "R23" for a room or "Ralf" for a name. Also the single character "A" or a single emoji would be correct.
The new help text for the plugin version 1.12.0 is not correct "The name of the Wi-Fi network (SSID) of the MoodleBox. The string must be a minimum of 8 bytes and a maximum of 32 bytes."
Ralf

ralf-krause, 2018-08-27:
Also in the code comment this fact is not considered. The code allows from 1 to 32 bytes.
`# SSID setting.
# Validate new SSID. Replace it with 'MoodleBox' if invalid.
# At this point, $NEWSSID is a string of hex values, e.g. "74657374" for "test"
# We want to check that it is valid, and between 8 and 32 bytes.
[[ $NEWSSID =~ ^([0-9a-fA-F]{2}){1,32}$ ]] || NEWSSID="4d6f6f646c65426f78" # "MoodleBox"
# New SSID is now valid; set it in config file.
`

martignoni, 2018-08-27:
You're right indeed. I stupidly wrote the help text by reading the comment in the code :-(

Can you please [open an issue in the MoodleBox plugin project](https://github.com/moodlebox/moodle-tool_moodlebox/issues)?