moodlebox/moodle-tool_moodlebox #44: Detect correctly other Raspberry Pi models

martignoni, 2018-06-07:
Currently RPi Zero W aren't detected correctly. Blocked by #43.

martignoni, 2018-06-12:
Passing.

martignoni, 2018-06-12:
Closing as fixed.

closed 2018-06-12 by martignoni

ralf-krause, 2018-06-12:
Hi Nicolas,
do you think that the Raspberry Pi Zero W could be a platform for the MoodleBox? Or is there a different idea what you would do with the Zero W?
Ralf

martignoni, 2018-06-12:
Answer is yes, but I have to solve beforehand the way to make the Pi Wi-Fi chip behave simultaneously as a client AND an access point. See e.g.

- https://www.raspberrypi.org/forums/viewtopic.php?p=1223117#p1223163
- https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md
- https://github.com/peebles/rpi3-wifi-station-ap-stretch 
- https://albeec13.github.io/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/ 

I've already worked on it, but not gotten far enough yet.