moodlebox/moodle-tool_moodlebox #46: Font problem with Safari and macOS

ralf-krause, 2018-06-21:
Hi Nicolas,
I found that I get a font problem with Safari and macOS on the settings page (1). The source code (3) seems to be okay but there is missing an emoji on the settings page. I switched of the font family (2) in a website admin tool to see the result without this setting.
Ralf
[image]
[image]
[image]

ralf-krause, 2018-06-21:
There is no problems in Chrome and Firefox on macOS. The appearance is also correctly in iOS on my iPad and my iPhone. No problem in Android 7.0 on my tablet.

ralf-krause, 2018-06-24:
I added the following lines table#moodleboxstatus to the styles.css. I don't know if this is the correct font-family list for all devices to show all characters and all emojis.

```
table#moodleboxstatus {
    font-family: 'Segoe UI', Roboto, 'Helvetica Neue', Arial, sans-serif, 'Apple Color Emoji', 'Segoe UI Emoji', 'Segoe UI Symbol';
}

table#moodleboxstatus td.c0 {
    font-weight: bold;
/*    font-family: 'Helvetica Neue'; */
}

table#moodleboxstatus tr.subinfo td.c0 {
    font-weight: normal;
    padding-left: 3em;
}
```

martignoni, 2018-06-24:
Cannot reproduce on my macOS High Sierra (10.13.5) platform. The ghost emoji was added recently to Apple platforms.

- What's  your OS version?
- Can you see 👻 displayed correctly here, and on this website too https://emojipedia.org/ghost/?

ralf-krause, 2018-06-24:
Hi Nicolas,
I 'm using macOS 10.13.5 ind Safari 11.1.1 (13605.2.8). I think that this is not the problem. I can see all ghosts on the page https://emojipedia.org/ghost/? ... most of the ghost are pictures only but the top 5 are characters.

As I  've shown on my screenshots the page has a font family list for the MoodleBox settings
`font-family: -apple-system,BlinkMacSystemFont,"Segoe UI",Roboto,"Helvetica Neue",Arial,sans-serif,"Apple Color Emoji","Segoe UI Emoji","Segoe UI Symbol";`
apple-system does not have the emojis  ... and when I delete -apple-system from this list then I get the emojis.
Ralf

martignoni, 2018-06-24:
> I 'm using macOS 10.13.5 ind Safari 11.1.1 (13605.2.8).

This is indeed not the problem, but the cause of the bad display. In your system, the "-system-font" CSS rule doesn't work (has no emojis), as can be seen on your first screen-shot (the font-family is not in red), hence the bad display. Boost uses by default the "-system-font" for main display. You may fix this in adding a rule in the Boost Raw SCSS box in Moodle GUI, instead of hacking the CSS files.

Summing up: this is a not a bug of my plugin, but it's a problem of the combination of your macOS/Moodle version. Probably even a Bootstrap issue. I suggest you open an issue on the Moodle tracker or on Boostrap, as Boost is based on it.

Closing as won't fix.

ralf-krause, 2018-06-24:
Thank you for your answer. I added an issue "Font problem with bootstrap and Safari on macOS".  https://tracker.moodle.org/browse/MDL-62788