moodlebox/moodle-tool_moodlebox #48: Enable setting of Wi-Fi regulatory country

martignoni, 2018-07-11:
Wi-Fi regulatory country is displayed in the plugin page, but can only be changed by editing file `/etc/hostapd/hostapd.conf`.

ralf-krause, 2018-07-12:
I try to set the country from DE to FR, the ssid from MoodleBox to MoodleBox7 and the channel from 1 to 7. I can see the country in /etc/hostap/hostap.conf.
[image]
[image]

ralf-krause, 2018-07-13:
Hi Nicolas, 
it works fine now. All settings are written into the /etc/hostapd/hostapd.conf.

To get the new settings into the settings list on the top of the Moodle page the admin must reload the page. Would it be possible to reload the settings list automatically?
Ralf

martignoni, 2018-07-13:
> To get the new settings into the settings list on the top of the Moodle page the admin must reload the page. Would it be possible to reload the settings list automatically?

Already tried, but not easy to do. Anyway that should be another issue.

martignoni, 2018-07-23:
Tested. Works correctly. Closing.

martignoni, 2018-07-31:
Reopening, as documentation is not updated yet.

ralf-krause, 2018-08-01:
Hi Nicolas,
yes I know that the country setting ist for the Regulation of rights. In US its Not allerdings to use channel 12 and 13. But in US it‘s allowed to use much higher transmitter power. 
Nobody will say in Germany that je ist setting the „WLAN-Regulationsland“. Everybody should use „Germany“ he is in Germany and he should only use „United States of America“ when he is in the USA. 
So everybody will know what we mean  with „WLAN-Länderkennung“. And I want to get it easy. Perhaps it could be an idea to Write about the regulatory in the knowledge base. 
Ralf