moodlebox/moodle-tool_moodlebox #53: The help text for SSID uses a wrong number of bytes

ralf-krause, 2018-08-27:
Hi Nicolas,
in this issue you fix the problem that nobody could enter a ssid with only one character. This is correct. You can enter any ssid such as "R23" for a room or "Ralf" for a name. Also the single character "A" or a single emoji would be correct.
The new help text for the plugin version 1.12.0 is not correct "The name of the Wi-Fi network (SSID) of the MoodleBox. The string must be a minimum of 8 bytes and a maximum of 32 bytes."

Also in the code comment this fact is not considered. The code allows from 1 to 32 bytes and this is correct.
`# We want to check that it is valid, and between 8 and 32 bytes.`
`[[ $NEWSSID =~ ^([0-9a-fA-F]{2}){1,32}$ ]] || NEWSSID="4d6f6f646c65426f78" # "MoodleBox"`

Ralf

ralf-krause, 2018-08-27:
I think that the help text could be something like this-
The name of the Wi-Fi network (SSID) of the Moodlebox can be from 1 to 32 bytes long. Please remember that some characters are using more than 1 bytes, e.g. Emoji.

martignoni, 2018-08-28:
Closing as fixed.

closed 2018-08-28 by martignoni