moodlebox/moodle-tool_moodlebox #96: Enable to set and change the fixed IP via GUI

martignoni, 2021-07-07:
See [this forum discussion](https://discuss.moodlebox.net/d/252-probleme-acces-page-web-apres-linstallation-de-limage).

martignoni, 2021-07-10:
Where to change the MoodleBox static IP address:
- `/etc/dhcpcd.conf`
- `/etc/dnsmasq.conf` (multiple times, dhcp range too)
- `/etc/hosts`
- `/etc/nodogsplash/nodogsplash.conf`

ralf-krause, 2021-07-11:
Hi Nicolas,
it's a great idea to allow the setting of the basis ip address for wifi. 

Will you allow to set every ip address? I don't know what would happen if the admin tries to set 46.38.232.227 because he found that this is the ip address for moodlebox.net.

The admin should only set a private ip address. Private ip addresses are 10.0.0.0/8, 172.16.0.0/12 and 192.168.0.0/16.

Perhaps it would it be better to preset 10.x.0.1 where the admin can set x from 0 up to 63. 
Also a preset of 172.x.0.1 with x from 16 up to 31 and a preset of 196.168.x.1 with x from 0 up to 255 could be options.

I'm looking forward for this useful feature!
Ralf

martignoni, 2021-07-11:
@ralf-krause Indeed, my first idea was to allow private addresses only. But I'll investigate your idea to restrict it to 10.0.0.0/8 network. Thanks.

ralf-krause, 2021-07-11:
Hi Nicolas,
the basic ip address 10.10.0.1/24 for the MoodleBox wifi could make problems when the ethernet ip address is 10.0.0.1/8. In this case both DHCP servers can provide the same ip address in both networks, for example 10.10.0.100. I think that the range of the MoodleBox wifi must be completely different from the range of the ethernet.
Ralf

martignoni, 2021-07-11:
Yes but this is impossible to know in advance: ethernet IP address could be any private one. This must be checked by the user.

ralf-krause, 2021-07-12:
Yes, I know. The problem comes up when the DHCP server on the ethernet has the basic ip address 10.0.0.1. When this DHCP server uses the full range of ip addresses from 10.0.0.1 up to 10.255.255.254 for the ethernet then you can't set a wifi ip address starting with 10 for the MoodleBox. In this case the admin needs to set an ip address starting with 192 or 172.

ralf-krause, 2021-07-13:
Okay, keep it simple. The programmer can't know which settings will be needed and useful in different cases. 

The admin should be able to set every basic ip address for the wifi. If he does not write a correct ip address then the default ip address 10.0.0.1 and the mask 255.255.255.0. If the admin tries to set an ip address which does not work then it will be his own problem. The MoodleBox will get a wifi and an ip address. The users will be able to connect to the MoodleBox even the ip address can't be used correctly with an internet connection. For example, if the admin sets 46.38.232.227 as a wrong basis ip address then nobody will get the site moodlebox.net anymore ... and some other addresses starting with 46.38.232.x.

I looked at OpenWrt and I found no restriction for the ip addresses. The MoodleBox should do the same. Every basis ip address can be accepted. And if the admin enters no ip address or a wrong syntax then the ip address is reset to 10.0.0.1 automatically.

martignoni, 2021-07-13:
This is how I'm implementing it.

See branch https://github.com/moodlebox/moodle-tool_moodlebox/tree/WIP-096-master_change_IP if you want to follow the development.

ralf-krause, 2021-07-13:
Great! I can set all private ip addresses. And I get an error when I take no private ip address. 

What happens with the dhcp addresses in the MoodleBox wifi when I insert a number not equal 1 at the last place. For example 10.10.10.10. Will the dhcp server know that it must send ip addresses 10.10.10.x and x not equal 1. I think that the MoodleBox settings do not change the address in the current state of development.

ralf-krause, 2021-07-20:
I installed the new plugin version 2.10.0-dev into my MoodleBox 3.12.0. For the installing I used the Moodle GUI "Install plugin from zip file". I restarted the MoodleBox.

To test a new feature I set the static ip address to 10.10.10.1. The ip address is valid and it is a basic address for the dhcp configuration. I restarted the MoodleBox again. A connection via MoodleBox wifi did not work any more. I connected via ethernet cable. The MoodleBox has a static ip address 192.168.1.100 on lan and also the host name moodlebox.hox works fine.

I connected to the MoodleBox with my browser and looked at the MoodleBox dashboard. The static ip address is shown as 10.0.0.1 but I set 10.10.10.1. I opened the terminal to get the ipconfig.

```
moodlebox@moodlebox:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.100  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::dc1e:8c53:3e39:3e3e  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:0f:a2:ae  txqueuelen 1000  (Ethernet)
        RX packets 6292  bytes 748654 (731.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7967  bytes 7981652 (7.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 20  bytes 1652 (1.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 20  bytes 1652 (1.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.10.10.1  netmask 255.255.255.0  broadcast 10.10.10.255
        inet6 fe80::9e84:becf:6ac:70be  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:0f:a2:b0  txqueuelen 1000  (Ethernet)
        RX packets 2255  bytes 150691 (147.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2362  bytes 388590 (379.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

moodlebox@moodlebox:~ $
```
Okay, the wifi address seems to be 10.10.10.1. Now I should look at the configuration files.

```
moodlebox@moodlebox:~ $ sudo cat /etc/hosts
127.0.0.1	localhost
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters

10.0.0.1	moodlebox  mb
moodlebox@moodlebox:~ $ sudo cat /etc/dhcpcd.conf
# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
# Some non-RFC compliant DHCP servers do not reply with this set.
# In this case, comment out duid and enable clientid above.
#duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
[... 65 lines cut ...]
FirewallRuleSet preauthenticated-users {
}
FirewallRuleSet users-to-router {
    FirewallRule allow udp port 53
    FirewallRule allow tcp port 53
    FirewallRule allow udp port 67
    FirewallRule allow tcp port 22
    FirewallRule allow tcp port 80
    FirewallRule allow tcp port 443
}
GatewayName MoodleBox
GatewayAddress 10.10.10.1
RedirectURL http://moodlebox.home/
GatewayPort 2050
MaxClients 50
SessionTimeout 360
moodlebox@moodlebox:~ $
```

I found the error in the hosts file. To get my MoodleBox with the short name and the short url `mb` I added the short name in the hosts file. I think that your script can't identify my changed line so it does not replace the ip address. I deleted my short name and your ip replacement works fine. Great work! I need to restart the MoodleBox when I change the ip address. Is this correct?

What should I do now with my MoodleBox project? I need the short url for my url shortener yourls and my settings with the ibeacons. The ibeacons only accept urls with 16 characters so I need to shorten the Moodle urls. https://machmitnetz.de/mod/page/view.php?id=161

Best regasds,
Ralf

ralf-krause, 2021-07-20:
I testet the static ip addresses 10.10.10.1, 172.16.20.1 and 192.168.10.1 ... the script works fine. The ip address is replaced correctly in all of the four configuration files.

When I try to set the static ip address to 10.10.10.10 the I get no error message but the MoodleBox sets the static ip address to 10.10.10.1. I also tried 172.16.20.20 and got 172.16.20.1. To see another one I set 192.168.100.100 and got 192.168.100.1. Surely we need an information about this fact.

In no case I could change the static ip address with a mobile device. When I change the static ip address the MoodleBox wifi gets an invalid state. The mobile device looses the connection without the chance to reconnect ... and I can't restart the MoodleBox with the mobile device.

Ralf

ralf-krause, 2021-07-20:
Three registrated mobile devices from three different ip ranges  :-)
Only the iPhone was connected via MoodleBox wifi. I took the screenshot with the notebook pool002 but it was connected via ethernet in this moment.

[image]

martignoni, 2021-07-20:
> I think that your script can't identify my changed line so it does not replace the ip address.

Your're right, I'll fix this

martignoni, 2021-07-20:
> When I try to set the static ip address to 10.10.10.10 the I get no error message but the MoodleBox sets the static ip address to 10.10.10.1. I also tried 172.16.20.20 and got 172.16.20.1. To see another one I set 192.168.100.100 and got 192.168.100.1. Yes this is correct in all cases but we surely need an information about this fact.

This is intended. As you suggested, the MoodleBox IP address will always end with `.1` segment. This will obviously be mentioned doc, that will follow later :-)

martignoni, 2021-07-20:
> Three registrated mobile devices from three different ip ranges :-)

This is also normal, even if undesirable: the leases file is updated at the end of the lease time (4 hours in this case) when a device has not connected.

I'll try however to find a workaround to empty it when the Wi-Fi settings are updated.

ralf-krause, 2021-07-20:
Hmmm ... there must be an error now. 
I installed the new plugin version and the script does not change the configuration files anymore. 

I did not change the file /etc/hosts for my first test. The last line was still `10.0.0.1        moodlebox` with no additional hostname. There also was no space behind the word moodlebox. Is this correct for the normal /etc/hosts?

ralf-krause, 2021-07-20:
Okay. The ip replacement seems to be okay. 
I tested a lot of versions of the file /etc/hosts without an alias and with one and more aliases. I used different whitespaces (one space, more than on space, one tab, more than one tab, spaces and tab). The script works fine when my device is connected via ethernet.

Trying to change the static ip address with a mobile device I can't reconnect to the MoodleBox wifi anymore. I must restart the MoodleBox. I'm using a Raspberry Pi 4B. Is it possible that the network restart does not work automatically?

ralf-krause, 2021-07-20:
For the next test I did the following:
My iPhone was connected to the MoodleBox wifi.
I changed the static ip address in the MoodleBox dashboard using a computer connected via ethernet. 
On my iPhone I looked at the wifi settings. 
The connection to the MoodleBox wifi stopped in the same moment I pressed the button "Change Wifi settings".
I tried to reconnect but it does not work.
In the Terminal I connected to the MoodleBox and I tried to restart some network services.
I don't know if all of these commands are needed.
```
moodlebox@moodlebox:~ $ sudo systemctl daemon-reload
moodlebox@moodlebox:~ $ sudo systemctl restart hostapd
moodlebox@moodlebox:~ $ sudo systemctl restart networking
moodlebox@moodlebox:~ $ sudo systemctl restart dhcpcd
moodlebox@moodlebox:~ $ sudo systemctl restart dnsmasq
moodlebox@moodlebox:~ $ 
```
After these commands I could reconnect my iPhone to the MoodleBox wifi and got a connection with an ip address and a dns service.

martignoni, 2021-07-20:
Did you reboot the MoodleBox after updating the plugin? This is needed when e.g. git pulling it, otherwise the script won't start.

ralf-krause, 2021-07-20:
Yes, I did a lot of reboots. Every time I changed the static ip address I need to reboot to get a wifi connection again for the mobile devices.

martignoni, 2021-07-20:
Just updated the script and tested.

Works OK from mobile too. No need to reboot after changing the IP. Rebooting is only necessary after updating the plugin.

ralf-krause, 2021-07-20:
I just tested you new update. Yes, it works now. When I change the ip address on my iPhone then the wifi icon is going away for a moment. And then the wifi icon comes back after a few seconds. The MoodleBox wifi is connected again, now with the new ip address.

Once I got an error message. I think the message come from another app running in the background. I have a lot of apps not allowed to use the mobile data. And if one of these apps tries to connect to the internet just without the wifi the the error message comes up. While I took the screenshot the wifi icon already was back again.

[image]

martignoni, 2021-07-27:
Closing as fixed.

closed 2021-07-27 by martignoni