moodlebox/moodle-tool_moodlebox #103: Reimplement footer buttons

martignoni, 2021-12-09:


martignoni, 2022-04-28:
Related discussion on moodle.org: https://moodle.org/mod/forum/discuss.php?d=434012.

ralf-krause, 2022-08-01:
Hi Nicolas,
perhaps you will get help from the code for the Moodle plugin "Boost Union". https://github.com/moodle-an-hochschulen/moodle-theme_boost_union

I just tested the plugin "Boost Union" to get some advanced settings for a web site with Moodle 4.0.2. You can set the course width to 1200px, you can set a favicon and you can add a footline. Currently you can see the footline with a button on a test page https://elisl.org/test40/

Best regards, Ralf

ralf-krause, 2022-08-01:
Please look also at https://github.com/moodle-an-hochschulen/moodle-theme_boost_union/issues/6

martignoni, 2022-08-02:
@ralf-krause: thanks for your help, but this will not work, as it seems plugins cannot override renderers (only themes can do that).

The plugin footer with buttons should work for any (Boost-based at least) theme, so we must find a way to do it in the plugin.

commit 836d72bc (2022-08-02): Fix for footer buttons

- This will fix #103.
- We use `<pluginname>_before_footer()` instead of `<pluginname>_standard_footer_html`.
- Reference: https://tracker.moodle.org/browse/MDL-68564.
- See also: https://github.com/davosmith/moodle-navbuttons/commit/f257bdc700eb5194e4096dd57daf1ab79c847ea1#diff-8f5ff72ef1cf59199cfc4bbf16f2475d09e448601242f0d2c51bb94bd4aa1395R41-R46
commit 3a47ece9 (2022-08-02): Fix for footer buttons

- This will fix #103.
- We use `<pluginname>_before_footer()` instead of `<pluginname>_standard_footer_html`.
- Reference: https://tracker.moodle.org/browse/MDL-68564.
- See also: https://github.com/davosmith/moodle-navbuttons/commit/f257bdc700eb5194e4096dd57daf1ab79c847ea1
closed 2022-08-02 by martignoni