moodlebox/moodle-tool_moodlebox #105: Text strings for tool_moodlebox

ralf-krause, 2022-08-03:
Some text strings should be changed because the product is outdated.
The link to the micro usb power supply should be https://www.raspberrypi.com/products/micro-usb-power-supply/ and not the old one https://www.raspberrypi.org/products/raspberry-pi-universal-power-supply/ . I think also that we should add a link for the usb-c power supply.

In the german language pack I am using the following text:
`<p>Verwenden Sie bitte die offizielle Stromversorgung <a href="https://www.raspberrypi.com/products/micro-usb-power-supply/" target="_blank"> für Raspberry Pi 3B / 3B+ / Zero2W </a> bzw. <a href="https://www.raspberrypi.org/products/type-c-power-supply/" target="_blank"> für Raspberry Pi 4B </a>.</p>`

This is set in the following strings: `badpowersupply, undervoltagedetected, undervoltageoccurred`

---
Do we currently need the strings for the plugin? `rpi4onemb, rpi4twomb, rpi4fourmb`

These three text strings were added when you released the plugin version 2.2.0 for the Raspberry Pi 4B. Nobody will install the plugin version 2.2.0 or the MoodleBox image 3.0.0 anymore. Currently we have the plugin version is 2.13.0 and the MoodleBox image 4.3.0 (or 4.2.0 with 32bit). https://docs.moodle.org/dev/String_deprecation#How_to_deprecate_a_string

Best regards, Ralf

martignoni, 2022-08-03:
Will change the strings with a new power supply description.

As for the deprecated strings, they are indeed deprecated: 
```
➜  git blame lang/en/deprecated.txt
a9f050fb (Nicolas Martignoni 2020-05-19 12:02:59 +0200  1) badpowersupply,tool_moodlebox
ed1da95a (Nicolas Martignoni 2017-09-13 09:18:10 +0200  2) changepasswordsetting,tool_moodlebox
ed1da95a (Nicolas Martignoni 2017-09-13 09:18:10 +0200  3) changewifipassword,tool_moodlebox
ed1da95a (Nicolas Martignoni 2017-09-13 09:18:10 +0200  4) currentwifipassword,tool_moodlebox
4ece3097 (Nicolas Martignoni 2021-07-19 21:34:35 +0200  5) ipaddress,tool_moodlebox
abf621b7 (Nicolas Martignoni 2019-03-01 11:16:53 +0100  6) moodleboxinfo,tool_moodlebox
abf621b7 (Nicolas Martignoni 2019-03-01 11:16:53 +0100  7) moodleboxinfofileerror,tool_moodlebox
abf621b7 (Nicolas Martignoni 2019-03-01 11:16:53 +0100  8) moodleboxpluginversion,tool_moodlebox
ed1da95a (Nicolas Martignoni 2017-09-13 09:18:10 +0200  9) newwifipassword,tool_moodlebox
ed1da95a (Nicolas Martignoni 2017-09-13 09:18:10 +0200 10) nopassworddefined,tool_moodlebox
d9744674 (Nicolas Martignoni 2020-06-17 18:34:58 +0200 11) raspbianversion,tool_moodlebox
f8c4c741 (Nicolas Martignoni 2019-07-15 18:41:03 +0200 12) rpi3,tool_moodlebox
913c1c58 (Nicolas Martignoni 2019-07-13 21:48:14 +0200 13) rpi4onemb,tool_moodlebox
913c1c58 (Nicolas Martignoni 2019-07-13 21:48:14 +0200 14) rpi4twomb,tool_moodlebox
913c1c58 (Nicolas Martignoni 2019-07-13 21:48:14 +0200 15) rpi4fourmb,tool_moodlebox
ed1da95a (Nicolas Martignoni 2017-09-13 09:18:10 +0200 16) wifipasswordmessage,tool_moodlebox
b3a2df10 (Nicolas Martignoni 2018-08-16 08:47:05 +0200 17) wifipasswordonhelp,tool_moodlebox
b3a2df10 (Nicolas Martignoni 2018-08-16 08:47:05 +0200 18) wifipasswordsetting,tool_moodlebox
```

But I could indeed delete some of them now, time flies…

closed 2022-08-04 by martignoni

ralf-krause, 2022-08-05:
I updated the german strings.

<p><b>Achtung: Die Spannung ist zu niedrig!</b> Die Stromversorgung der MoodleBox ist unzureichend. Dies könnte verschiedene Probleme verursachen, z.B. die Anzahl der WLAN-Clients begrenzen oder die MoodleBox unerwartet herunterfahren.</p>
<p><b>Prüfen Sie Ihr Netzteil.</b> Verwenden Sie möglichst das <a href="https://www.raspberrypi.com/products/micro-usb-power-supply/" target="_blank"> Raspberry Pi 12,5 W Micro-USB-Netzteil </a> für Raspberry Pi 3A+, 3B, 3B+, Zero W und Zero 2 W oder das <a href="https: //www.raspberrypi.com/products/type-c-power-supply/" target="_blank"> Raspberry Pi 15 W USB-C-Netzteil </a> für Raspberry Pi 4B.</p>

<p>Seit dem letzten Start der MoodleBox wurde mindestens einmal eine zu niedrige Spannung registriert. Möglicherweise ist die Stromversorgung der MoodleBox unzureichend. Dies könnte verschiedene Probleme verursachen, z.B. die Anzahl der WLAN-Clients begrenzen oder die MoodleBox unerwartet herunterfahren.</p>
<p><b>Prüfen Sie Ihr Netzteil.</b> Verwenden Sie möglichst das <a href="https://www.raspberrypi.com/products/micro-usb-power-supply/" target="_blank"> Raspberry Pi 12,5 W Micro-USB-Netzteil </a> für Raspberry Pi 3A+, 3B, 3B+, Zero W und Zero 2 W oder das <a href="https: //www.raspberrypi.com/products/type-c-power-supply/" target="_blank"> Raspberry Pi 15 W USB-C-Netzteil </a> für Raspberry Pi 4B.</p>

commit 93bb5fb0 (2022-08-04): Update links to correct products

- Fixes #105.