moodlebox/moodle-tool_moodlebox #108: Incorrect execution rights in the plugin folder

ralf-krause, 2022-08-05:
I just found that the cron job does not run since my installation of the plugin tool_moodlebox 2.13.0-beta.

The page http://moodlebox.home/admin/index.php?cache=1 shows: 
The script admin/cli/cron.php has not been executed for 4 days 3 hours, but should run every 1 minute.

Executing the cron command I get an error
```
moodlebox@moodlebox:~ $ sudo nice -n10 ionice -c2 /usr/bin/php /var/www/moodle/admin/cli/cron.php
Moodle upgrade pending, cron execution suspended.
moodlebox@moodlebox:~ $ 

```

Okay. 4 days ago I installed tool_moodlebox 2.13.0-beta. In the meantime I have installed the plugin versions 2.13.0 and 2.13.1. I also updated Moodle to Moodle 4.0.2+ (20220805). Now I am searching the error cause.

I found that the execution rights are not set in the folder bin
```
moodlebox@moodlebox:/var/www/moodle/admin/tool/moodlebox/bin $ ls -al
total 36
drwxr-xr-x 2 www-data moodlebox  4096 Aug  5 12:04 .
drwxr-xr-x 9 www-data moodlebox  4096 Aug  5 12:04 ..
-rw-r--r-- 1 www-data moodlebox  2123 Aug  5 12:04 changepassword.sh
-rw-r--r-- 1 www-data moodlebox 10619 Aug  5 12:04 changewifisettings.py
-rw-r--r-- 1 www-data moodlebox  1086 Aug  5 12:04 changewifisettings.sh
-rw-r--r-- 1 www-data moodlebox   819 Aug  5 12:04 pijuicestatus.py
-rw-r--r-- 1 www-data moodlebox  1022 Aug  5 12:04 resizepartition.sh
moodlebox@moodlebox:
```

Did I make a mistake in my installation? 
To install plugins I go to http://moodlebox.home/admin/tool/installaddon/index.php and upload the zip file to the upload field. Then I click on the button “Install plugin from ZIP file”. Could it be that the execution rights are changed by this installation script?

martignoni, 2022-08-05:
I've suffered this once, when upgrading the plugin to an intermediate version. It's certainly caused by the beta version, which had not incremented the version number.

It has nothing to do with permissions.

To fix this, try to log into your MoodleBox and attempt an upgrade via CLI:
```
sudo -u www-data php /var/www/moodle/admin/cli/upgrade.php
```
This should fix the problem.

ralf-krause, 2022-08-05:
```
moodlebox@moodlebox:~ $ sudo  /usr/bin/php /var/www/moodle/admin/cli/cron.php
Moodle upgrade pending, cron execution suspended.
moodlebox@moodlebox:~ $ sudo -u www-data php /var/www/moodle/admin/cli/upgrade.php
Für die installierte Version 4.0.2+ (Build: 20220805) (2022041902.04) ist kein Upgrade notwendig. Danke für den Versuch!
moodlebox@moodlebox:~ $ sudo  /usr/bin/php /var/www/moodle/admin/cli/cron.php
Moodle upgrade pending, cron execution suspended.
moodlebox@moodlebox:~ $ 
```

ralf-krause, 2022-08-05:
Okay, I deinstalled the plugin tool_moodlebox and I hope to get the cron running without it. No, it does not work,

martignoni, 2022-08-05:
Did you reboot?

ralf-krause, 2022-08-05:
Yes, I did.

I hope I got a solution for this issue .... see https://moodle.org/mod/forum/discuss.php?d=212888#p1266648

```
moodlebox@moodlebox:~ $ sudo su
root@moodlebox:/home/moodlebox# /usr/bin/php /var/www/moodle/admin/cli/cron.php
Moodle upgrade pending, cron execution suspended.
root@moodlebox:/home/moodlebox# /usr/bin/php /var/www/moodle/admin/cli/maintenance.php --enable
== Wartungsmodus (http://moodlebox.home) ==
Ihr System ist derzeit im CLI-Wartungsmodus. Ein Zugriff aus dem Internet ist nicht möglich.
root@moodlebox:/home/moodlebox# /usr/bin/php /var/www/moodle/admin/cli/maintenance.php --disable
== Wartungsmodus (http://moodlebox.home) ==
Der Wartungsmodus wurde beendet. Die Website ist wieder verfügbar.
root@moodlebox:/home/moodlebox# /usr/bin/php /var/www/moodle/admin/cli/cron.php
Server Time: Fri, 05 Aug 2022 18:31:33 +0200
...
...
Cron script completed correctly
Cron completed at 18:31:36. Memory used 44.4 MB.
Execution took 3.169385 seconds
root@moodlebox:/home/moodlebox# 
```

Now I could install the plugin tool_moodlebox version 2.13.1 again. I moved the zip file into the tools folder, unzipped it and set the owner to www-data:moodlebox. Then I rebooted and then I must do the same again. Now it seems to work.

```
root@moodlebox:/home/moodlebox# cd /var/www/moodle/admin/tool/moodlebox/bin
root@moodlebox:/var/www/moodle/admin/tool/moodlebox/bin# ls -al
total 36
drwxr-xr-x 2 www-data moodlebox  4096 Aug  4 18:25 .
drwxr-xr-x 9 www-data moodlebox  4096 Aug  4 18:25 ..
-rwxr-xr-x 1 www-data moodlebox  2123 Aug  4 18:25 changepassword.sh
-rwxr-xr-x 1 www-data moodlebox 10619 Aug  4 18:25 changewifisettings.py
-rwxr-xr-x 1 www-data moodlebox  1086 Aug  4 18:25 changewifisettings.sh
-rwxr-xr-x 1 www-data moodlebox   819 Aug  4 18:25 pijuicestatus.py
-rwxr-xr-x 1 www-data moodlebox  1022 Aug  4 18:25 resizepartition.sh
root@moodlebox:/var/www/moodle/admin/tool/moodlebox/bin# 
```

The execution rights are correct. The Moodlebox setting can be set. I tested the ip address, channel, and the regulation country.

martignoni, 2022-08-06:
Very likely a problem of version number. Closing.

Please reopen if needed.

closed 2022-08-06 by martignoni

ralf-krause, 2022-08-06:
I found that the MoodleBox does not execute the script admin/cli/cron.php. I get this message on the system notification page http://moodlebox.home/admin/index.php?cache=1

But the log file /var/log/syslog shows that the cron runs every minute. And when I start the admin/cli/cron.php  manually from terminal then it shows an normal output without any error. When the admin/cli/cron.php runs manually the message still is shown on the system notification page.

Only when I clear the caches in Site Administration > Development > Clear caches then the message goes away. What is wrong with my MoodleBox after the reinstallation of the MoodleBox plugin?

martignoni, 2022-08-06:
Probably nothing. Just a problem with the intermediate beta version.

martignoni, 2022-08-06:
Sorry, I was wrong: this is a bug introduced in Moodle 4.0.2 (not in MoodleBox, neither in MoodleBox plugin). See [my announcement in MoodleBox forum](https://discuss.moodlebox.net/d/329-bug-in-moodle-402-prevents-proper-operation-of-moodlebox).

martignoni, 2022-08-06:
Workaround meanwhile:
1. Open http://moodlebox.home/cache/admin.php
2. Under "Configured store instances", edit "TMPFS application" and "TMPFS session" stores, emptying totally their field "Cache path". Don't forget to save the changes.