moodlebox/moodle-tool_moodlebox #126: Wrong dashboard view after changewifisettings

ralf-krause, 2023-11-26:
I am testing MoodleBox 4.6.0. I found that I get a wrong dashboard view after I changed the MoodleBox ip address. 

I set the regulation country to Germany and the static ip address to 10.10.10.1. My connected laptop md-lap-3011 gets an ip address 10.10.10.131 … this is correct. The laptop wifi settings show this ip address and the correct router ip address 10.10.10.1. The MoodleBox dashboard still shows the static ip address 10.0.0.1. There is also shown an older ip address for Tab-S6-Lite-von-Ralf which should be deleted when the ip address changed.

[image]
[image]

martignoni, 2023-11-28:
The bug with regulatory country is fixed in version v.2.17.2 of the plugin.

martignoni, 2023-12-02:
See #128 for progress on item 3 in [above comment](https://github.com/moodlebox/moodle-tool_moodlebox/issues/126#issuecomment-1828311182).

martignoni, 2023-12-24:
Closing, since the related issues are fixed and released.

closed 2023-12-24 by martignoni