moodlebox/moodle-tool_moodlebox #131: Bug: The MoodleBox dashboard shows a wrong time

ralf-krause, 2024-02-22:
The first time I saw the problem was when I setup the MoodleBox 4.6.0 to use an RTC. 
Please look at https://discuss.moodlebox.net/d/247-using-a-real-time-clock-module-on-moodlebox/8
The date/time string in the MoodleBox dashboard is different to the computer time. Yes, I thought there could be a problem with my RTC settings.

Today I did a new test. I installed a fresh microSD card with the MoodleBox 4.6.0 image. I started the MoodleBox with the new image. The MoodleBox was connected to the router so the date/time string should be correct. Yes, the date/time string in the dashboard is different to the computer time again. In my addional test I connected via ssh to get the system date/time from the Terminal ... the system date/time is correct but not the string in the MoodleBox dashboard.

You can see my tests in in the following screenshot. There are the computer time, the terminal, and the MoodleBox dashboard.

Ralf

[image]

ralf-krause, 2024-02-22:
I forgot to write ... I changed nothing with the MoodleBox 4.6.0 image. Everything comes out of the box.

ralf-krause, 2024-02-22:
I installed MoodleBox 4.5.1 on another microSD card to look at the same things in MoodleBox 4.5.1. In this MoodleBox version the dashboard shows the correct date/time string.

[image]

martignoni, 2024-02-22:
This is not really a bug, since one can (and should) set the correct default timezone for the MoodleBox under Moodle, via __Site Administration > Location settings__.

Beware that the date depends also from the timezone setting of each user, so if you've changed the admin user's timezone in the Moodle profile, the date displayed in the dashboard is updated accordingly.

Anyway, the default timezone for PHP (and Moodle) is currently "Europe/London", which is not consistent with the OS default timezone, that is "Europe/Zurich", so this should be fixed. Thank you for the report.

Sidenotes:
- The change of behaviour between v4.5.1 and v4.6.0 is certainly due to PHP update to 8.2.
- This has nothing to do with your RTC, since a RTC has to be on UTC time.