moodlebox/moodle-tool_moodlebox #139: Problem with the button osition

ralf-krause, 2024-08-09:
Testing the new version of tool_moodlebox I found that on some pages the additional MoodleBox support buttons are switching into the top line. You can see this on the following pages. 
http://moodlebox.home/admin/index.php?cache=1
http://moodlebox.home/admin/plugins.php
http://moodlebox.home/admin/environment.php

I can‘t say If we can see the problem only on admin pages. 

I 'm using a Raspberry Pi 4B (4GB RAM), Moodle 4.4.1 (20240809), and the MoodleBox tool 2.18.0 (2024080700).

[image]

PatLemaire, 2024-08-09:
Which browser do you use? According to the screenshot, the problem is found on a mobile application?

ralf-krause, 2024-08-09:
I get this problem with all devices. I tested: 

- an iPad with Safari, Chrome and Firefox
- my iPhone with Safari, Chrome and Firefox
- an tablet with Android 14, Chrome and Firefox
- my iMac 27" with macOS 13.6.9, Safari, Firefox, Chrome, Brave
- Linux PC mit Chrome and Firefox

ralf-krause, 2024-08-11:
I tested some other things. I set the debugging message to DEVELOPER and I got no warnings from the MoodleBox tool and no warnings from other plugins.

In the MoodleBox settings I activated also the option "Show date and time setting in footer" and both button options were placed on the top of some admin pages. I tested if the option "Restart MoodleBox" really restarts it ... yes, it does.

ralf-krause, 2024-11-02:
How does the plugin put in the buttons to the page footer? 
Is there a process which is waiting for the end of every page?
Is there a timeout which shows the buttons while some admin pages running to long?

martignoni, 2024-11-03:
> How does the plugin put in the buttons to the page footer? Is there a process which is waiting for the end of every page? Is there a timeout which shows the buttons while some admin pages running to long?

See function [`before_footer_html_generation`](https://github.com/moodlebox/moodle-tool_moodlebox/blob/483560d2ff48e0a1d7a63289d176898fda9eb142/classes/hook_callbacks.php#L43).

ralf-krause, 2024-11-05:
The footer buttons are placed at the top of some admin pages also in the new MoodleBox 4.8.0. I tested this with a fresh installed MoodleBox so no one of my normally installed plugin can be the problem.

[image]

PatLemaire, 2024-11-05:
With the release of Moodle 4.4, some functions have been deprecated. We've had to rewrite some of the code to ensure compatibility, but placement is a problem. I'm going to see if manipulating the DOM in javascript might allow more efficient (stable) placement.

PatLemaire, 2024-11-07:
For the moment, investigations show that the buttons are not positioned correctly on all pages where the _$output->footer()_ function is not used as usual.
Let's quote https://moodledev.io/docs/4.4/apis/subsystems/output:
```
$PAGE->set_url($url);
$PAGE->set_title($title);
$PAGE->set_heading($title);

$output = $PAGE->get_renderer('tool_demo');

echo $output->header();
echo $output->heading($pagetitle);

$renderable = new \tool_demo\output\index_page('Some text');
echo $output->render($renderable);

echo $output->footer();
```
You seem to have listed the 3 pages concerned (others are intended to be called up indirectly).
A closer look at the code in admin/index.php, [lines 823-824](https://github.com/moodle/moodle/blob/99b5729824524e0bdc6a64ff1b0e4423d1d4b37a/admin/index.php#L823):

>     echo $OUTPUT->header();
>     echo $OUTPUT->footer();
Content of the page is build after footer call!

martignoni, 2025-04-28:
Issue still here with version 2.20.0 of the plugin and Moodle 5.0.