moodlebox/moodle-tool_moodlebox #145: No wifi connection with image 4.6.0 on Raspberry Pi 3B model

martignoni, 2024-10-27:
See https://discuss.moodlebox.net/d/428-moodlebox-460-connexion-wifi-impossible.

ralf-krause, 2024-11-01:
This issue is only fixed when the parameter wifi-sec.proto is set when starting the MoodleBox. The parameter should be set to wpa (for WPA) on a Raspberry Pi 3B and to rsn (for WPA2) on the newer models. 

A microSD card working for a Raspberry Pi 3B can be moved to a Raspberry Pi 4 but the wifi only has WPA and not WPA2. Currently a microSD card working for a Raspberry Pi 4B with WPA2 would not work for a Raspberry Pi 3B without WPA2.

martignoni, 2024-11-01:
Reopening. See discussion in #150.

ralf-krause, 2024-11-01:
Now I installed the MoodleBox 4.5.1 image in a microSD card .... this is the last MoodleBox image with Bullseye.

I started the Raspberry Pi 3B with this microSD card and it works! The MoodleBox wifi uses the security protocol WPA2. So it could not be the hardware. It must be something wrong in the software. Does WPA2 have parameters for the crypto method or the crypto depth.

martignoni, 2024-11-01:
> Now I installed the MoodleBox 4.5.1 image in a microSD card

This bug is present in Debian Bookworm only (MoodleBox image 4.6.0 or later).

ralf-krause, 2024-11-01:
Yes, I wanted to be sure that the Raspberry Pi 3B can be an access point with WPA2. So I tested MoodleBox 4.5.1 with Debian Bullseye.

martignoni, 2024-11-01:
See https://github.com/moodlebox/moodlebox/issues/354 for a definitive fix.

closed 2024-11-01 by martignoni